arduino/avrdude-build-script #7: include solution for Adafruit Trinket for Linux/Ubuntu

facchinm, 2017-06-09:
_Moved from https://github.com/arduino/Arduino/issues/6360 by @jerabaul29_

Hi!

It is well known that the Adafruit Trinket (based on ATtiny85) has problems with the Arduino IDE on Linux/Ubuntu. For my part, trying to upload code from the Arduino IDE, I get:

```
An error occurred while uploading the sketch
avrdude: error: usbtiny_transmit: Input/output error
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: error: usbtiny_transmit: Input/output error
```

But applying the solution offered here solved the problem, and I am able to upload code successfully now:

https://andijcr.github.io/blog/2014/07/31/notes-on-trinket-on-ubuntu-14.04/

The solution is basically a change in some definitions for the ATtiny in the avrdude.conf:

https://gist.github.com/andijcr/f4a660fde4035fb0a3aa

Would it be possible to make the arduino IDE automatically apply those changes when dealing with the Trinket? If not, what is preventing you from doing this? I ask here but maybe I should see with the Adafruit library, just tell me and I will open an issue there in this case ;)

matthijskooijman, 2017-06-11:
I haven't looked at the changes in detail, but did anyone report this as a bug or patch upstream with avrdude yet? That sounds like the right place to fix it, so I think Arduino should be hesitant to patch this in their avrdude version, and only do that as a temporary measure when upstream avrdude has merged or at least reviewed the patch.

facchinm, 2017-06-12:
@matthijskooijman  totally agree, I didn't have time to check the patch yet but, if relevant, it should be forwarded to Joerg :slightly_smiling_face:

facchinm, 2018-10-03:
Should be fixed by avrdude-6.3.0-arduino14

closed 2018-10-03 by facchinm