arduino/avrdude-build-script #11: Arduino as ISP programmer can't be used with avrdude 6.3.0-arduino12

per1234, 2018-03-30:
Using Windows 10, 64 bit, Arduino IDE 1.8.5 and 1.9.0-beta build 38

1. File > Preferences > Additional Boards Manager URLs > https://downloads.arduino.cc/packages/package_avr_3.6.0_index.json > OK
1. Tools > Board > Boards Manager > install Arduino AVR Boards 1.6.206 > Close
1. Tools > Programmer > Arduino as ISP
1. Tools > Burn Bootloader

Verbose output:
```
C:\Users\per\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino12/bin/avrdude -CC:\Users\per\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino12/etc/avrdude.conf -v -patmega328p -cstk500v1 -PCOM14 -b19200 -e -Ulock:w:0x3F:m -Uefuse:w:0xFD:m -Uhfuse:w:0xDE:m -Ulfuse:w:0xFF:m 

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\per\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino12/etc/avrdude.conf"

         Using Port                    : COM14
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 Version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: stk500_initialize(): n_extparms 6 mayor 1 minor 18 
avrdude: stk500_initialize(): n_extparms = 6
avrdude: sending extparam for arduino multi isp
avrdude: stk500_set_extended_parms(): can't get into sync
avrdude: stk500_initialize(): failed
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: stk500_disable(): protocol error, expect=0x14, resp=0x12

avrdude done.  Thank you.

Error while burning bootloader.
```
`-v -v -v -v` output for the command above:
https://gist.github.com/per1234/e3a8349d7bb188fc0c315e6393e6a84a

I am running **File > Examples > 11.ArduinoISP > ArduinoISP** from Arduino IDE 1.8.5 on the Arduino as ISP programmer.

The programmer works fine with avrdude 6.3.0-arduino9.

The problem also occurs for Upload Using Programmer with the Arduino as ISP:
https://gist.github.com/per1234/c4fa5cf3e12ce5aa2ddfaeeb5d4f1a26

I can use USBasp, AVRISP mkII, and USBtinyISP programmers with avrdude 6.3.0-arduino12 with no problems.

Originally reported at: http://forum.arduino.cc/index.php?topic=538183.msg3669159#msg3669159

WestfW, 2018-04-15:
Avrdude is sending 7 bytes of "Extended parameters":
   avrdude: Send: E [45] . [07] . [04] . [d7] . [c2] . [00] . [00] . [00]   [20] 
The Arduino ISP software is only prepared to "ignore" 4 parameters:
https://github.com/arduino/Arduino/blob/5f86bdd1b4fc1677dbd8cbd466dafc39e0619287/build/shared/examples/11.ArduinoISP/ArduinoISP/ArduinoISP.ino#L658
While the STK500v1 protocol theoretically allows a variable number of parameters (it includes a length byte), I don't think "stk500v1" should be getting updates at this point, so I would classify this as a bug in avrdude.  (however, a workaround in ArduinoISP (or whatever) should also be possible.

WestfW, 2018-04-16:
Grr.  avrdude is now apparently pulled from https://github.com/facchinm/avrdude rather than from the official avrdude sources, and that repository has added additional parameters (reset_pin and reset_pin_second_board ??) to the stk500 programmer code, IF it reports "minor SW version 18" (which is what arduinoISP uses.)
https://github.com/facchinm/avrdude/issues/1

facchinm, 2018-04-17:
@per1234 @WestfW thanks for the report. We are using https://github.com/facchinm/avrdude  as a "replacement" until official avrdude makes an actual release.
That commit slipped in, sorry for that; it was meant to be used internally and not to cause any problem; I'm rebuilding the package and publish it later today.

per1234, 2018-04-17:
Fixed in Arduino AVR Boards 1.6.207/avrdude 6.3.0-arduino14.

closed 2018-04-17 by per1234

jvabole, 2019-10-31:
AVRdudess works with arduino as an isp programmer if set by Atmel AVRISP and 19200 baudrate