arduino/avrdude-build-script #15: Please add upstream Status Update to README

E3V3A, 2019-02-12:
Is this repo still getting syncing/patching with upstream?
I'm trying to find a properly and maintained *github* mirror or patch repo for avrdude. 

Yes, I blood refuse to use that stupid savannah svn repo!
And who in their right mind is still using mailing lists for bug tracking and issue reports!??

facchinm, 2019-02-13:
This repo (https://github.com/facchinm/avrdude/commits/master) is not really syncing but was 1-to-1 with svn about one year ago (last commit from mainline is patch #8444).
On the top of that there are some patches we use on the arduino build (https://github.com/arduino/avrdude-build-script/blob/e22771dd9fed8edbcffb3e9f06d0a44aab467c01/avrdude-6.3.build.bash#L23)

E3V3A, 2019-02-13:
Just to clarify. 

1. Are you representing a/the Arduino developer(s)? and 
2. will you be maintaining this for future arduino ISE builds?

I'm trying to understand where investing my dev time will not be wasted on some soon-to-be-dead repo.

facchinm, 2019-02-13:
1. yes :slightly_smiling_face: 
2. I expect to move the repo to `arduino` organization sooner or later (it surely won't disappear anyway)

E3V3A, 2019-02-13:
Great! :) 

> I expect to move the repo to arduino organization sooner or later (it surely won't disappear anyway)

Huh? I though this **is** under the arduino org...

facchinm, 2019-02-13:
I was referring to the other one ([https://github.com/facchinm/avrdude/](https://github.com/facchinm/avrdude/)) :slightly_smiling_face:

E3V3A, 2019-02-13:
Lol, yeah, its very confusing.

I sincerely believe that it is the fault of insisting to use that outdated and hard to maintain savannah repo. People wanna develop on github, and thus there are literally [hundreds of](https://github.com/search?q=avrdude) avrdude related repos. Each with some small hack etc. 

It's a very sad situation for this very important software. In addition there are tons of [proposed patches](http://savannah.nongnu.org/patch/?group=avrdude) (**currently ~50**), that are either ignored or not merged. So with a closing rate of ~5 per/year, no wonder the dev situation is disastrous! :disappointed: 

***We need ONE place for avrdude, that works!***  
(I hope Brian S. Dean  read this...or whoever is in charge of that place nowadays!)

matthijskooijman, 2020-12-29:
@facchinm, it's a bit confusing that this repo has some patches, but the avrdude.git it uses (Under your account) *also* seems to have Arduino-specific patches (rather than just being a clean git-copy of the SVN repo). Even more, some (probably all) of the patches in this repo seem to be commited to that repo as well. What's up with that?

mirh, 2023-05-10:
An update on the status of this repo (and avrdude-packing) wouldn't hurt.

umbynos, 2023-05-10:
The relevant patches have been ported to https://github.com/arduino/avrdude-packing/tree/main/patches or [have been upstreamed/replaced](https://github.com/avrdudes/avrdude/pulls?q=is%3Apr+is%3Aclosed+author%3Aumbynos). The plan is to archive this repo and use only [avrdude v7.1-arduino.1](https://github.com/arduino/avrdude-packing/releases/tag/7.1-arduino.1) in the avr and megaavr cores. The only thing missing is to run a lot of testing, since the changes from avrdude were a lot from 6.3 to 7.1... If you want to help with testing you can use avrdude v7.1-arduino.1

MCUdude, 2023-05-19:
> If you want to help with testing you can use avrdude v7.1-arduino.1

MightyCore, MiniCore, MegaCore, MicroCore, MajorCore, and MegaCoreX have all been using [avrdude v7.1-arduino.1](https://github.com/arduino/avrdude-packing/releases/tag/7.1-arduino.1) for several months now, and there has not been reported a single issue where Avrdude 7.1 didn't work where 6.3 did. PlatformIO has also bundled [avrdude v7.1-arduino.1](https://github.com/arduino/avrdude-packing/releases/tag/7.1-arduino.1).

So the v7.1-arduino.1 release is currently being used and tested by thousands of users every single day. I'd argue that it's safe to replace the 6.3 version with 7.1 in all applications.

Avrdude v7.2 will probably be released this summer. It provides lots of under-the-hood improvements, and a greatly enhanced terminal mode (-t). However, I'd argue that you should bundle 7.1 first, and let us 3rd party Arduino core bundle the latest and greatest releases, to make sure they are stable and bug-free.