arduino/avrdude-build-script #25: Atmega4809 Curiosity Nano upload not working under Linux

Szum123321, 2021-05-14:
Hi

I've recently bought the Microchip Atmega4809 Curiosity Nano dev board rev 5. Sadly, uploading doesn't work on Linux.
The error message says: 
```
avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/[user]/.arduino15/packages/MegaCoreX/hardware/megaavr/1.0.8/avrdude.conf"
         User configuration file is "/home/[user]/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping
         Using Port                    : usb
         Using Programmer              : curiosity_updi
avrdude: usbdev_open(): Found nEDBG CMSIS-DAP, serno: ATML3094051800000641
avrdude: usbdev_open(): WARNING: failed to set configuration 1: Device or resource busy
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
avrdude: usbdev_send(): wrote -5 out of 912 bytes, err = Input/output error
avrdude: jtag3_edbg_prepare(): failed to send command to serial port
```

The issue occurs on both Arch Linux 5.12.3-arch1-1 and Linux Mint 20.04(i think). I tested both flatpak version 1.8.13 and the normal nightly build 1.8.14 2020.05.12 ide. I also did the test on Windows with ide 1.8.13 and everything worked.

The same issue has been discussed [here](https://github.com/MCUdude/MegaCoreX/issues/13), on MegaCoreX github.

leo60228, 2021-06-24:
Have you found a solution for this?

leo60228, 2021-06-24:
This also affects the AVR128DB48 Curiosity Nano (which is expected, seeing as they use the same debugger).

leo60228, 2021-06-24:
I've done some troubleshooting and these lines seem unrelated to the core issue:
```
avrdude: usbdev_open(): WARNING: failed to set configuration 1: Device or resource busy
```
```
[ 6502.418058] usb 1-3: usbfs: interface 0 claimed by usbhid while 'avrdude' sets config #1
```
They can be worked around by forcing usbhid, usb-storage, and cdc-acm to avoid claiming the device. This requires a kernel patch for the latter, but can be done via kernel flags for the former two. This doesn't seem to do anything other than remove the error message, however.

leo60228, 2021-06-24:
Turns out these can be programmed by copying a .hex file to an emulated mass-storage device, which bypasses avrdude completely.

Szum123321, 2021-07-08:
For now I installed newest version of avrdude from AUR (avrdude-svn) and replaced binary file in `~/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino18/bin/avrdude` with a symlink to /usr/bin/avrdude